sandialabs/NuMAD #7: Compatibility with openFAST and ElastoDyn

ahmedhassan554, 2022-03-22:
Hi, I want to use Sandia's SNL61p5m for 5MW baseline in openFAST. When I changed blades in elastoDyn from "NRELOffshrBsline5MW_Blade" to "FASTBlade_precomp", the following error comes up
"Invalid numerical input for file. unable to read NB1InpSt".
I have two questions.
1. Is openFAST compatible with these files e.g FASTBlade_precomp?
2. If not, what is the solution so I can use this blade configuration for my simulation?

mayankchetan, 2022-03-22:
Hi ahmedhassan554,

I think the error you are facing is due to a file format mismatch. This is common when various tools by different sources are in rapid development. 

Assuming you are using the most recent version of [OpenFAST (v3.1.0)](https://github.com/OpenFAST/openfast/releases/tag/v3.1.0) and referring to the [NREL 5MW r-test blade](https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/5MW_Baseline/NRELOffshrBsline5MW_Blade.dat) file as an example, you can solve this by commenting out:

1. [Line 25 of writeFastBlade.m](https://github.com/sandialabs/NuMAD/blob/8267c457f8bb50df6f888bcca2e6b6b45fca797a/source/utilities/designTools/fast/writeFastBlade.m#L25) file. 
2. [Line 30 of writeFastBlade.m](https://github.com/sandialabs/NuMAD/blob/8267c457f8bb50df6f888bcca2e6b6b45fca797a/source/utilities/designTools/fast/writeFastBlade.m#L30) file. 

Please let me know if this solution works for you.

p.s I am not a developer of NuMAD nor a part of Sandia National Labs, but have extensively used the older version of NuMAD (v2) in my research and would like to contribute back to the community.

ahmedhassan554, 2022-03-23:
Dear Mayank,
Yes, it does work!!. Thank you so much. 
Just a side note that you can delete directly the 1st and 6th line from the .dat file.

ecamare, 2022-03-30:
Thanks for your help @mayankchetan