sandialabs/NuMAD #11: Error running example for Excel2NuMADObject

robynnemurrayNREL, 2022-05-17:
Hi there,

I am getting the following errors when I try to run the excel to NuMAD object example. Can you please help :) 

NuMAD and Design Code path setup complete.
Error using AirfoilDef (line 52)
Could not open file "airfoils/DU99-W-405.txt"

Error in StationDef (line 44)
                    obj.airfoil = AirfoilDef(af);

Error in BladeDef/addStation (line 148)
                obj.stations(k) = StationDef(af);

Error in xlsBlade (line 103)
        blade.addStation(affile,afspan(k));

Error in excel2NumadObject (line 9)
blade = xlsBlade(designFile);

robynnemurrayNREL, 2022-05-17:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
>
> CAUTION: This email originated from outside of NREL. Do not click links or open attachments unless you recognize the sender and know the content is safe.
>
> Hi Robynne,
>
> Thanks for using our code! What branch are you on? If you are on "dev" can you run the example from the "main" branch?
>
> Ernesto

Thanks for your fast response. I am using the main branch 😊

Also, unrelated – but in NuMAD 2.0 excel sheet you could define the number of layers of each material at each spanwise section. In the new 3.0 excel sheets, is this no longer the case?

Thank you so much!

rclarke17, 2022-05-17:
Hey Robynne,
For the original question, check to see if your airfoils folder matches the contents of the airfoil folder in the github repository. If it does and you're still having problems then I would pull a clean version of the main branch again, I was able to run the ExcelToNumadObject example script earlier today. 

As for the question about changes to the excel sheet between the NuMAD object (v3.0) and NuMAD GUI (v2.0). The NuMAD Object excel script now has a tab called Components, in this tab there are two columns called 'CP span' and 'CP nLayers'. 'CP span' defines at what span your layers change and 'CP nLayers' defines how many new layers of that component material there is.

Best,
Ryan

robynnemurrayNREL, 2022-05-20:
Thank you Ryan!

It's nice to hear from you! How are you enjoying work at SNL? I hope you are doing well!

I got the example model working, and have been able to run the script with my own excel input file. I'm actually modeling the SNL MHKF1 tidal turbine blade!

This example doesn't spit out the precomp output files, is there an easy way to get it to?

Also, how can I change the element size for the ANSYS output?

Thank you!

rclarke17, 2022-05-20:
Good to hear from you too Robynne!
You can generate precomp files if you uncomment out line 15, the line "blade.generateBeamModel".

robynnemurrayNREL, 2022-05-23:
Hi again Ryan,

Our blades don't have shear webs so I am getting errors when I try to run precomp. I can play with it and add shear webs with material thicknesses of 0, but is there a better way to go about this?

This is the error it gives me.

Thank you!
***@***.***

rclarke17, 2022-05-23:
I'm not seeing what error you are getting, can you try sending it again?

robynnemurrayNREL, 2022-05-23:
Can you see the attached screen shot?
***@***.***

closed 2022-07-05 by rclarke17