sandialabs/NuMAD #16: How do you install this Software???

marty121412, 2022-07-05:
I have a windows x64 I cannot for the life of me to work out how to install this software please help? 
I have ANSYS & MATLAB downloaded onto my machine.

ecamare, 2022-07-07:
Hi Marty, Sorry to hear the trouble. Have you followed the directions on our documentation [()?](https://numad.readthedocs.io/en/latest/user/getting_started.html#install-numad) If so, what step are you on? In general, there are two ways to install it. One way is to "clone" the repository with Git and the other is a direct download of the Matlab files.

rclarke17, 2022-07-07:
Hi Marty,
NuMAD v3.0 has new documentation which can be found in the link below.
http://numad.readthedocs.io/

marty121412, 2022-07-07:
Thanks Ryan I'll check it out

Kind regards,

Marty

http://numad.readthedocs.io/<https://nam12.safelinks.protection.outlook.com/?url=http%3A%2F%2Fnumad.readthedocs.io%2F&data=05%7C01%7C%7C8dae4a943aa4403210ef08da603bb21a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637928106806271503%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=vNZ7eCReohfGIRxDDjv5m9h0Db5lb7wmZoBlRB6QYMM%3D&reserved=0>

ecamare, 2022-07-07:
Hi Marty, 

No, I did not send you a link for NuMAD 2.0. The link I provided was is for NuMAD 3.0. Ryan's link is for the documentation in general. My link points you to the installation portion specifically.

Best,
Ernesto

marty121412, 2022-07-07:
Hi Ernesto,

The link you sent looks like just what I'm after thanking you for your assistance.

Kind regards,

Marty

GncDude, 2022-07-20:
Hello Sir/s, I cant setup. I am sorry I couldnt not understand… I have student edu Ansys and Matlab, anaconda and python and visualstudio. Mat you show us setup tutorial video please? Please :( King Regards

closed 2022-12-20 by ecamare