sandialabs/NuMAD #17: problem about NuMAD v2

ali-rouhbakhsh, 2022-08-04:
when i try to open a simple model this error occured:

Error using cell2mat (line 52)
CELL2MAT does not support cell arrays containing cell arrays or objects.

Error in NuMAD_main>loadmodel (line 911)
    set(cell2mat(app.smdp.hg_sm),'Visible','off');

Error in NuMAD_main>openmodel (line 815)
    loadmodel(cbo,[]);  % finish loading required data and display model

Error in NuMAD_main>cb_openmodel (line 725)
        openmodel(pn,fn,cbo);
 
Error while evaluating Menu Callback.

ecamare, 2022-08-11:
Sorry for the delay. Is this still a problem for you? How are you trying to open it?

ali-rouhbakhsh, 2022-08-12:
Yes, I still have a problem
After I installed NuMAD, I opened it through MATLAB and from the open model section, I tried to open one of the sample models included by the software called SNLWindPACTv1.3.

rclarke17, 2022-09-26:
have you copied the file txt2mat.m to your MATLAB search path?

closed 2022-12-20 by ecamare