sandialabs/NuMAD #18: help

fpxno1, 2022-09-21:
could you tell me how to transform the blade.nmd model into ansys FEM model,I am a graduate student majoing in mechanical engineering

rclarke17, 2022-09-26:
There are two ways to generate an ANSYS model from an .nmd file. If you have a blade object generated that corresponds to your .nmd blade you can use the new NuMAD v3 way using the command:

layupDesign_ANSYSmesh(blade, 'name of your .nmd file goes here')

For an example of this look at the ExcelToObject folder in examples.

The old NuMAD v2 way to generate an ANSYS model from an .nmd file is to open the .nmd file in the NuMAD GUI, to access the GUI type numad into Matlabs command line after adding the Numad paths using the command addNumadPaths, then once you are in the GUI go file-> open model and select your .nmd file, once you have loaded in your .nmd file you can go to ANSYS in the tool bar and select Generate Now to build an ANSYS model.

omaudhuit, 2022-09-29:
@rclarke17 0 - Hope all is well. Was wondering if I could get some help on some errors I am getting when running the excel2NumadObject.m script. 

I can't seem to be able to generate the mesh. 

[image]

rclarke17, 2022-09-29:
Try replacing line 19 with this:
BladeDef_to_NuMADfile(blade,NuMADfile,'MatDBsi.txt', strcat(pwd, '\airfoils'));

omaudhuit, 2022-09-30:
@rclarke17 - Thank you very much for the response. 

Trying to get more familiar NuMAD since we have mainly been working with FAST but would like to start gaining a deeper understanding on the structural side. 

Could you share some insight into the mesh generation - is this contained within the .src file that is outputted from the above script? The readthedocs for v3.0 doesn't go into so much detail regarding how ANSYS is interfaced. 

Any feedback would be much appreciated.

rclarke17, 2022-09-30:
My guess is that the number of nodes and elements it is trying to make exceeds the allowable limit of ANSYS student version. 
one thing you could try is using a coarser mesh, by setting blade.mesh in the Excel2NumadObject.m script to 0.5, this will make a very poor mesh but it's just a check of if that is the issue. 

As for your question of how NuMAD interfaces with ANSYS, it is through the shell7.src script, essentially NuMAD translates the geometry data of the wind blade into ANSYS APDL batch language that are then written to the shell7 script, we then rely on ANSYS to mesh the geometry for us.

One of our development tasks that we at Sandia are working on right now is to decouple blade mesh creation from ANSYS, essentially we are creating an in-house mesher for NuMAD so that any FEA program can then be used to analyze your blade.

omaudhuit, 2022-10-03:
@rclarke17 - Thanks again for the feedback and the information.

The Ansys student version may be the cause. I've even tried reading the SNL61p5m_NuMAD_121212 reference model .src file directly in ADPL but have come across a series of errors (with the program crashing entirely at times):

[image]

If you have ever come across such behavior do let me know. 

I will have to look into acquiring another Ansys license.

fpxno1, 2022-10-07:
whether the numad 3.0 has more specifical explanation,I want to learn more about it

omaudhuit, 2022-10-07:
@fpxno1 - Not sure if this is what you are looking for but I would definitely start with the available documentation: https://numad.readthedocs.io/en/latest/user/fea.html

fpxno1, 2022-10-07:
> @fpxno1 - Not sure if this is what you are looking for but I would definitely start with the available documentation: https://numad.readthedocs.io/en/latest/user/fea.html

thank you bro,I have found that,and I successfully run those examples,but i don't know how to use NUMAD according to need,I am a student exploring the wind turbine blade area without tutor's guidance ,It is too difficult, wish to  communicate more.

omaudhuit, 2022-10-10:
> @rclarke17 - Thanks again for the feedback and the information.
>
> The Ansys student version may be the cause. I've even tried reading the SNL61p5m_NuMAD_121212 reference model .src file directly in [...]

@rclarke17 - Following up on this to see whether or not you or a colleague of yours have ever come across any issues even with the shell.src file from the reference model. Any response will be helpful. Thank you.

ecamare, 2022-12-20:
Th ANSYS error "An element referred to as 91 ..."  occurs because newer versions of ANSYS discontinued the SHELL91 element. We have fixed this problem and are closing this issue. Please let us know if you still need help.

veliemresavas, 2024-08-03:
> Th ANSYS error "An element referred to as 91 ..." occurs because newer versions of ANSYS discontinued the SHELL91 element. We have fixed this problem and are closing this issue. Please let us know if you still need help.

I do still encounter this issue while using the src in apdl?

yunzhiliuliu, 2025-02-06:
Just my observation: I tried using GUI (open .nmd file) to run the example ExcelToObject. The generated .src file still has shell element 99 and 91 before line57. 

[image]

[image]

> Th ANSYS error "An element referred to as 91 ..." occurs because newer versions of ANSYS discontinued the SHELL91 element. We have fixed this problem and are closing this issue. Please let us know if you still need help.