sandialabs/NuMAD #24: Error exporting shell7.src ANSYS input file

ncepulibei, 2022-11-03:
I encountered the following problems when exporting the shell7.src ANSYS input file. My model seems to check without problems. Is it because I did not set different thicknesses for each blade segment?

Incorrect use of chckxy (line 51)
Data sites should be different from each other.
Error pchip (line 58)
[x,y,sizey] = chckxy(x,y);
Error resampleAirfoil (line 82)
pp = pchip(t,xy');
Error write_ shell7 (line 67)
af_ out = resampleAirfoil(station.coords, round(Nansys/2-1), 'arc');
Error NuMAD_ main>cb_ generate (line 3411)
write_ shell7(app,blade,filename);
Error calculating Menu Callback.

ncepulibei, 2022-11-04:
I think I have found the reason for this problem. I have several airfoils with flat trailing edges. According to the prompt, I need to add DPs at all corners, but when adding DPs at the trailing edge, the points on the upper surface cannot be added, which shows the following problem：
the index is out of the array range.
Error NuMAD_ main>cb_ modifySMD_ uictrls (line 2406)
app.station(k). sm{end+1} = app.station(k).sm{n};
Error computing UIControl Callback.
So I adopted the following method. I changed the coordinates of the points to 99.9%, and then I changed them to 100% after adding successfully, so that I can pass the check blade data, But I think it is this hidden danger that leads to my inability to generate the ansys model, so how can I solve it?

ncepulibei, 2022-11-04:
I have solved this problem. It is the problem of the coordinate file of the flat trailing edge airfoil

ecamare, 2022-11-04:
Okay. Glad you solved the problem.

closed 2022-12-20 by ecamare