sandialabs/NuMAD #26: the result of bpe and the precomb

ncepulibei, 2022-11-21:
When I calculate the blade section characteristics with bpe and precomb respectively, the results differ greatly. Why? In addition, when running bpe, polyfitdata will appear Mat does not exist. What type of file is it? Do you have case files? I will attach my kout file and a section file, input_ 1.out_ Gen, according to the comparison with the results in other papers, the precomb calculation results are more accurate, while the result of bpe is larger, which is why
[kout.txt](https://github.com/sandialabs/NuMAD/files/10057018/kout.txt)
[input_1.out_gen.txt](https://github.com/sandialabs/NuMAD/files/10057125/input_1.out_gen.txt)

rclarke17, 2022-11-21:
There are some general issues with BPE, I would trust Precomps results.

ncepulibei, 2022-11-21:
But I want to generate (6 × 6) Is Precomb unable to achieve the stiffness matrix of? What should we do? I also want to change the direction of the ply, so that there will be bending torsional coupling. There is no (6 × 6) If the stiffness matrix of the blade can not be simulated? Is there any good solution

rclarke17, 2022-11-21:
Neither Precomp nor BPE are adequate at generating off-axis stiffness coupling terms. For off-axis terms ANBA4 or VABS are your best solutions. Here's a link to a paper discussing them and a preprocessor for them called SONATA.
https://www.sciencedirect.com/science/article/pii/S0263822320326817

closed 2022-12-20 by ecamare