sandialabs/NuMAD #31: Error using BPE function: "

ZL6462, 2023-03-28:


ZL6462, 2023-03-29:
Dear @rclarke17,

I have check the output.txt, may be every thing is OK, the file is attached as follows:
[output.txt](https://github.com/sandialabs/NuMAD/files/11099785/output.txt)

Besides, I also try running "shell7bpe.src" manually through the APDL GUI, and there is a warning as follows, and I an not sure whether this warning matters.
[image]

Thank you for your attention!

kbonney, 2023-07-11:
@ZL6462 For us to help you with this problem please provide a minimum example which replicates the issue you are encountering. This should include any data and code that are not included in NuMAD.

rclarke17, 2023-07-11:
ZL6462,
BPE was discontinued with NuMAD v2.0. If you want to use it you need to use NuMAD v2.0.
As for the warning message that appears when manually running shell7bpe.src, you can ignore that.

kercc1985, 2023-10-21:
Hi，@ZL6462 ，
Have you solved this problem? I have encountered the same issue, do you have a solution?  thanks.

ZL6462, 2023-12-18:
> ZL6462, BPE was discontinued with NuMAD v2.0. If you want to use it you need to use NuMAD v2.0. As for the warning message that appears when manually running shell7bpe.src, you can ignore that.

Dear @rclarke17 ,

Do you mean that NUMAD-3.0 doesn't support BPE function?

Thank for your attention