sandialabs/NuMAD #35: sphinxcontrib-matlabdomain 0.19.0 pre-release

joeced, 2023-04-28:
Hi

I'm the main author and maintainer of the `sphinxcontrib-matlabdomain` Python package.
I just finished a major update of the extension and added a new configuration option:
`matlab_short_links`. In short (no pun intended), it's finally possible to render and link to MATLAB classes and function by the name they have in MATLAB.

Instead of `src.package.AmazingClass`, you now get `package.AmazingClass`. Further, the linking to that is `` :class:`package.AmazingClass` ``. Or in your case rather than `matlab.get_pet_metadata`, it would become just `get_pet_metadata`

Secondly, we updated the class rendering to be more MATLAB like, with summaries of constructor, properties and methods.

As you are a major user of the extension, I would appreciate if you could take the time to test if the new release works as expected. In your `requirements.txt`, please specify:

```
sphinxcontrib-matlabdomain==0.19.0rc1
```

If you need any help or discover issues please reach out to me in this thread.

Thanks!

joeced, 2023-05-16:
I now released the version after testing it on a larger project.

```
sphinxcontrib-matlabdomain==0.19.0
```

kbonney, 2023-05-22:
Hi @joeced. Thanks for reaching out. Sorry we were not able to assist in testing the latest RC. We have actually been having issues building our documentations recently. I believe it might have to do with the matlabdomain extensions - I'll be reaching out on your repo soon about it.

closed 2023-05-22 by kbonney