sandialabs/NuMAD #37: Where is the script "layupDesign_ANSYSmesh.m", "layupDesign_ANSYSanalysis.m"

namnguyen-lqdtu, 2023-05-26:
In the user's guide at this link: https://numad.readthedocs.io/en/latest/user/fea.html, some scripts are mentioned (for examples: "layupDesign_ANSYSmesh.m", "layupDesign_ANSYSanalysis.m") but they dont exist in the source code. Please provide those codes with example of using them.

namnguyen-lqdtu, 2023-05-26:
Thank you @ecamare! 
I have downloaded the NuMAD-dev code and found the scripts. I am now  trying to run the scripts in the "example" folder. For the example in the "runIEC" folder, when running the scripts "IECLoadsAnalysis.m", the line 33 doesn't work. It goes to Debugging mode and stops at the line 68 in the function "resampleAirfoilDB", waiting for an input from keyboard (see pictures below). What do I have to do now?
[image]
[image]

kbonney, 2023-07-11:
@namnguyen-lqdtu Seems like you are encountering a remnant of some debugging code. This is something we should address, but in the mean time I believe you can get around this by just pressing the continue button located in the top toolbar.

moonstar123456, 2023-09-18:
> Thank you @ecamare! I have downloaded the NuMAD-dev code and found the scripts. I am now trying to run the scripts in the "example" folder. For the example in the "runIEC" folder, when running the [...]

Hello, Nice to meet you! i am learning numad 3.0 and have some questions, is it possible to consult you? My email is 843526493@qq.com, skype: emilymoonjiang.   Many thanks in advance!

namnguyen-lqdtu, 2023-09-18:
Yes. You can come back to me in October. I am too busy in September.Sent from my iPhone

Hello, Nice to meet you! i am learning numad 3.0 and have some questions, is it possible to consult you? My email is ***@***.***, skype: emilymoonjiang.   Many thanks in advance!