sandialabs/NuMAD #42: How about a python version?

mcMunich, 2023-06-12:
Hey Sandia, I am interested in using numad again (I used it for research) but would like to see if brought over to python. Any chance that happens in the near future?

ecamare, 2023-06-12:
Yes, we have been working on converting numad to python for about one year already. Going forward, we will leave the MATLAB version as a static copy and will only develop in python. You can access our current work in the pynumad-dev branch. Note however, the documentation is not yet complete so using it might be a bit difficult. We had a WESC presentation about this last month.

kbonney, 2023-07-11:
The Python implementation of NuMAD now lives on a standalone repository at https://github.com/sandialabs/pyNuMAD. It is still in early stages, but active use will help move development. Please feel free to try the Python version  and give us feedback.

closed 2023-07-11 by kbonney

mcMunich, 2023-08-23:
Cool, when I get a chance I will do some testing. M