sandialabs/NuMAD #49: Error using cell2mat

kercc1985, 2023-10-19:
When I run Numad in the way of M files and create a new model, the following error occurs. I hope that friends who have encountered similar problems and successfully solved them can help me solve them.

Error using cell2mat (line 52)
CELL2MAT does not support cell arrays containing cell arrays or objects.

Error in NuMAD_main>loadmodel (line 911)
set(cell2mat(app.smdp.hg_sm),'Visible','off');
...

kercc1985, 2023-10-19:
Hi，@kbonney 
Thank you for your response. Now I will describe the specific situation to see if it can help solve the problem:

I installed NuMAD according to the Userguide, and the specific situation is as follows:
1.The Matlab version is R2016a.
2.I copied snl.m and txt2mat.m to Matlab's search path, C:\Users\kercc\Documents\MATLAB, and modified the path in snl.m to point to the source files of NuMAD: SNL_NuMAD_Toolbox_path = 'D:\Matlab_workspace\NuMAD-2.0\source'.
3.I entered "snl" in the Matlab command line, and it displayed "SNL NuMAD Toolbox v13.04.03 path setup script complete."
4.I entered "numad" in the Matlab command line, and the NuMAD Interface was launched.
5.When I clicked on "file > new model", the following error popped up：

Error using cell2mat (line 52)
CELL2MAT does not support cell arrays containing cell arrays or objects.

Error in NuMAD_main>loadmodel (line 911)
    set(cell2mat(app.smdp.hg_sm),'Visible','off');

Error in NuMAD_main>cb_newmodel (line 700)
    loadmodel(cbo,[]);  % finish loading required data and display model
 
Error while evaluating Menu Callback

The above is the entire process, and no models or examples have been run.

bkloeg, 2023-11-22:
Hi @kercc1985 and @kbonney,
I experience exactly the same issues. I can manage to open the NuMAD interface but when I want to create a new file or open an existing one, I get the error that you describe. 
Do you have any update on this topic?

kercc1985, 2023-11-22:
> Hi @kercc1985 and @kbonney, I experience exactly the same issues. I can manage to open the NuMAD interface but when I want to create a new file or open an existing one, I get the error that you describe. Do you have any update on this topic?
Hi，bkloeg，
In fact, I didn't find a good solution in NuMad2.0. Later, I tried Numad3.0 and found that the problem didn't exist.

bkloeg, 2023-11-23:
@kercc1985 Okay, I will try to set up NuMAD 3.0 as well then. Do you have any tips for installation? I have downloaded the zip and followed the steps in Anaconda prompt and Matlab but I don't know how to launch the program.