sandialabs/NuMAD #52: Generate Abaqus input files

MitchellH141, 2024-01-25:
Hello, 

Apologies if this is not the usual way to contact you, I am very new to using NuMAD. 
I was wondering if there was anyway to generate input files (e.g. .inp) to generate FE models in Abaqus rather than ANSYS.

Many thanks, 
Hannah

rclarke17, 2024-01-25:
Hi @MitchellH141 ,
NuMAD 3.0 does not have an Abaqus capability. However, we have been working on and just released a new NuMAD code written in Python called pyNuMAD, which can be accessed here: https://github.com/sandialabs/pyNuMAD/tree/main
Which does have scripts that can generate an ABAQUS mesh for you. The one current drawback to pyNuMAD is we don't have automated analysis scripts built for it currently, so any analysis you will want done you will have to setup yourself.

MitchellH141, 2024-01-26:
Thank you very much for the link. 
I have been following to user guide but for some reason I cannot seem to import pynumad. It has successfully installed and I have no errors to suggest that there was an issue with the installation but when I try to import pynumad I get a FileNotFoundError indicating that the software_paths.json file cannot be found. Is this an error in the way that I have installed pynumad? 
Many thanks.

For some reason I can't upload an image to this comment but the error is:
FileNotFoundError: [Errno 2] No such file or directory: 'c:\\Miniconda3\\envs\\engd\\Lib\\site-packages\\pynumad\\software_paths.json'

ecamare, 2024-01-26:
Hi Hannah, 

Does it give a file name and/or line number where it is trying to read the software_paths.json?

MitchellH141, 2024-01-26:
Hi Ernesto, 

It does, the file is c:\Miniconda3\envs\engd\Lib\site-packages\pynumad\io\excel_to_blade.py:9

and the line where the error happens is line 8 I think:
**8** with open(join(package_dir, "software_paths.json"), "rt") as f:

Apologies, I still can't seem to be able to attach an image otherwise I'd copy the error. 

Thanks very much,
Hannah

kbonney, 2024-01-26:
Hi Hannah,
The issue is that the software_paths.json file gets left out of the installation when you run `pip/conda install .`. We haven't configured the package to solve this problem properly, so the solution is to install by running `pip install -e .` in the root directory of the repository. The `-e` does a "local" installation which means that the package runs directly out of the files you clone through git, so the software_paths.json is available.

MitchellH141, 2024-01-26:
Hi Kirk, 
That solved it, thanks very much!

closed 2024-01-29 by ecamare