sandialabs/NuMAD #54: SNL61p5m.nmd project issues

dntjudme, 2024-07-18:
Hi all, 

Does anyone encounted similar issues as I do?

As you open the SNL61p5m.nmd project in the NuMAD GUI and direct to the 'Modify skin material divisions', it is found not able to change the division points on the viewing station. 
Or one can also try to modify the '% Chord', but I guess you won't see anything being changed actually. 

Additionally, I found for the XLS-to-NuMAD utility, it just not work for the SNL61p5m.nmd project as it always pops out the error message: 'xls2nmd encountered an error', and it overwrites the SNL61p5m.nmd project file. 

The above error seems only occurs when operate with the SNL61p5m.nmd project, for other blade models, the NuMAD works just fine.

Can anyone suggested how to fix this? I need to work with the SNL61p5m.nmd model since my research is dedicated for the NREL 5MW baseline wind turbine blade. 

Thanks!
Zs
[image]
[image]

gilberto-melo-goca, 2025-04-28:
Hi, you can send to me the file in nmd ?

gilberto-melo-goca, 2025-05-09:
Hello, I would like the SNL61p5m.nmd file. Could you send it to me? I am doing my final college work and I wanted to simulate this blade in several tests.